shaise/FreeCAD_FastenersWB #326: FreeCAD v0.21.2 Object of type FeaturePython is not JSON serializable

CallThatAKnife, 2024-01-31:
Hi and hope you're well. Loving the WB and thankyou for developing it! It's been a lifesaver for my projects.

I'm no coder but am seeing a new error come up after every save of a design that uses the FastenersWB. It recently started happening in v0.21.0 and after the upgrade to v0.21.2 I'm seeing the same behaviour.

The report view gives me the following:
`11:13:49 PropertyPythonObject::toString(): failed for <class 'FastenersCmd.FSViewProviderTree'>
11:13:49 Traceback (most recent call last):
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps
return _default_encoder.encode(obj)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode
chunks = self.iterencode(o, _one_shot=True)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode
return _iterencode(o, 0)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default
raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
11:13:49 PropertyPythonObject::toString(): failed for <class 'FastenersCmd.FSViewProviderTree'>
11:13:49 Traceback (most recent call last):
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps
return _default_encoder.encode(obj)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode
chunks = self.iterencode(o, _one_shot=True)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode
return _iterencode(o, 0)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default
raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
11:13:49 PropertyPythonObject::toString(): failed for <class 'FastenersCmd.FSViewProviderTree'>
11:13:49 Traceback (most recent call last):
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps
return _default_encoder.encode(obj)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode
chunks = self.iterencode(o, _one_shot=True)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode
return _iterencode(o, 0)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default
raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable`

Any thoughts or easy fixes I can implement here?

Cheers again and please let me know if I can improve my posting; I'm a virgin around these parts.

shaise, 2024-01-31:
Hi,

I'm using FreeCAD v0.21.2 and have no issues saving fasteners.

Need from you the following:
1. open freecad, press help->about freecad, press "copy to clipboard" and paste the info here
2. start with a clean document, add a fastener and see if saving works. If not, please tell me which fastener caused that.
3. if the above works, please give me the simplest scenario that can cause this problem so I can reproduce it.

shai.

CallThatAKnife, 2024-01-31:
Thanks for getting back so quickly Shaise.

Some additional information for you as well as this may have contributed (maybe a corrupt project file?):

I'm running 3 monitor, a 24" in portrait with all the report and other "informational" views and 2x27" in landscape with the design window in one and the spreadsheet in the other. When FreeCAD starts, the views on the 24" monitor always need to be moved up as I'm assuming they're respecting the vertical resolution/position information of the 27" monitors. The first time I saw this error was after moving the views on the 24" during startup where FreeCAD froze then shut down. Perhaps soemthing has gone wrong here? It started in v0.21.0 so I reinstalled with v0.21.2 and noted the same behaviour.

For your numbered points:

1. [code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * ExplodedAssembly
  * fasteners 0.4.58
  * freecad.gears 1.0.0
  * Manipulator 1.5.0
  * parts_library 2022.10.28
  * ProDarkThemePreferencePack 1.0.0
  * Render 2023.2.6
  * ThreadProfile 1.87.0
[/code]

2. In setting up for this one, when I moved my combo view window up on the 24" monitor I got the following in the report view: 14:40:51  QMainWindowLayout::tabPosition called with out-of-bounds value '0'. This repeats _sometimes_ when the view is moved again.

The save was successful; the error came again:
`14:43:03 PropertyPythonObject::toString(): failed for <class 'FastenersCmd.FSViewProviderTree'>
14:43:03 Traceback (most recent call last):
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps
return _default_encoder.encode(obj)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode
chunks = self.iterencode(o, _one_shot=True)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode
return _iterencode(o, 0)
File "C:\Users\User1\AppData\Local\Programs\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default
raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
`

3. Structure of file is Part - Body - Padded Sketch based around origin. Circle validated and set to diameter 100mm. Pad 10mm. Added ISO4762 M6x8-Screw and upon Save, error was produced. Does this happen for you too?

[image]

CallThatAKnife, 2024-01-31:
No worries; file attached.
[test.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/14116463/test.zip)

CallThatAKnife, 2024-01-31:
So just reopened the file I sent you and my report window is now on my primary 27" screen instead of the 24". No errors yet.

Nothing moved and saving exhibits the same behaviour with exactly the same error.

shaise, 2024-01-31:
Got your file. I can load and save it without any issues. I really do not know what triggers your problem.
you can try the freecad forum, perhapse someone else encounter this.
as a last resort you can start from clean - meaning deleting all data under c:\users\<username>\AppData\roaming\freecad, starting from default layout and reinstall all workbenches one by one.

CallThatAKnife, 2024-01-31:
Thanks for trying Shai and sorry to have taken up your time. I'll reinstall from fresh as you suggest and hopefully all will be well again. I think that somehow with the 24" running 1080p and the two 27s running 1440p in another orientation maybe the views are meeting some kind of out of bounds error that's propogating into what we see. It's not a showstopper (though let's see if the original design file is somehow corrupted!!)

I'll get to reinstalling and let you know if the error persists. Thanks again.

CallThatAKnife, 2024-01-31:
Hi again; just to close out.

I've reinstalled FreeCAD v0.21.2 and everything is working as it should be - including maintaining the view positions on the 24" monitor. Oddly enough, now PowerToys FancyZones won't let me automagically span FreeCAD across the two 27" monitors but manually no worries - and FreeCAD remembers this as well the next time I open it.

Sooo...nothing to see here I guess and I must thank you again for walking down this path to nowhere with me.

Take care.

closed 2024-01-31 by CallThatAKnife