shaise/FreeCAD_FastenersWB #334: [feature] ASME/ANSI B 18.6.1 wood screws

Anthony-Gaudino, 2024-02-07:
I was looking to add a `flat countersunk #12 wood screw`  and I didn't find it in FastenersWB.
These American standard screws are part of `ASME/ANSI B 18.6.1`.

### References:

https://www.scribd.com/doc/261941351/ANSI-B-18-6-1-1981-R1997
https://pdfcoffee.com/ansi-b-1861-1981-r1997-pdf-free.html

https://www.globalfastener.com/standards/detail.php?sid=MjY0Mw==
https://www.globalfastener.com/standards/detail_4560.html
https://www.globalfastener.com/standards/detail_4558.html

https://boltport.com/standards/asme-b18.6.1/

Anthony-Gaudino, 2024-02-07:
Data in Excel: [ansi_b_18_6_1.xlsx](https://github.com/shaise/FreeCAD_FastenersWB/files/14201910/ansi_b_18_6_1.xlsx)

When min and max values where given I only got min values.
Please let me know if max values are necessary.

cfunseth, 2024-02-09:
I'm working on several other ASME fasteners and can add this to my list if someone else doesn't do it before me.

cfunseth, 2024-02-09:
> When min and max values where given I only got min values. Please let me know if max values are necessary.

@shaise @alexneufeld @hasecilu Is there a preference from you all on the use of maximum, minimum, or average dimensions in this workbench? It looks like ISO usually clarifies that nominal = maximum, but I don't believe ASME specifies this. See below for examples:

ISO:
[image]

ASME:
[image]

hasecilu, 2024-02-09:
Usually when is not mentioned, I use the max dimensions

cfunseth, 2024-02-13:
@Anthony-Gaudino Initial support for ASME wood screws should be included now. It should show up if you update your workbench from the Addon manager in FreeCAD.

hasecilu, 2024-03-02:
I think this issue should be closed

shaise, 2024-03-03:
@Anthony-Gaudino, can we close this?