shaise/FreeCAD_FastenersWB #340: [Discussion] Implementing GUI translation on Crowdin.

kaktusus, 2024-02-13:
Hey.

@shaise  I can help you with running translations on the FreeCAD-addons repository in Crowdin.
It is not complicated and does not require much time.
I expect that this will make your workbench more widely used. 😄

shaise, 2024-02-13:
1. I am not familiar with Crowdin. Does this mean that you must be on line to make the translations work?
2. What about the translations we already have?

kaktusus, 2024-02-13:
ad. 1 To begin with, I can do everything by introducing you to the subject.
ad. 2 The translations that we have will not be lost.
Ts language files can be imported by anyone from the translator upwards.

---
I know people working with GUI translations who work this way.
They download the string file for translation. They modify it locally and then upload it to Crowdin.
This is nothing special.
It may sound scary to a person who has not yet performed such operations.

It is possible that when the translations will be on Crowdin there will be less work with them for the workbench maintainer. Now you have to validate every PR from every translator.
Import/export of translations comes down to a few clicks in the web interface. No scripts are required to import export.
Of course, after implementation, some regularity will be required for exports from Crowdin. More "disadvantages" I do not see.

---
At this stage it is important whether you agree that the translations are supported by the Crowdin platform.

Don't worry about the rest. You are not alone. :wink:

kaktusus, 2024-02-13:
[image]

This is what the panel for basic management of source translation files looks like.
To add a string file to the translations:
* the first time you click **add file**
* for each subsequent **update**

---

[image]

And this is what the basic panel for managing translation files looks like.
* here you can prepare a "zip package" with all languages or only selected ones.
* and download it to your computer.

---
Then what remains to be done is to compile the downloaded files into qm format and attach them to the sources of your workbench.

I once prepared a very versatile bash script for translators that prepares compiled translation files.
https://github.com/FreeCAD/FreeCAD-translations?tab=readme-ov-file#bash-script--2022-04-17

kaktusus, 2024-02-14:
Oh I overlooked. I didn't have much time yesterday when I was replying to you. :wink: 
After the whole procedure, the translations will be in the FreeCAD user's file system. As it was before.

Internet connection will be needed to:
* download the source code from GitHub to perform string extraction.
* upload the file with strings for translation to the Crowdin platform,
* download the translations prepared by the translators from Crowdin,
* prepare a PR with the translations.

Crowdin is very helpful for preparing translations by the international translation community.

Let's agree that I can help you with the management of translation files and with Crowdin operation. :wink: 
But it's worthwhile for you to acquire these skills as well. :1st_place_medal:  However, there is no rush and no one will force you to do it.

shaise, 2024-02-14:
Thanks, I will take a closer look and decide.

hasecilu, 2024-03-09:
From what I have seen on the main FreeCAD Crowdin project almost every day FreeCAD is being translated, on the addons project only 7 external WB are there from more than 100!.
The point is that having the same people that translates main FreeCAD translating the addons will help end users by having a level of consistency on the whole FreeCAD environment.

I'm sure doing this will help everyone.

kaktusus, 2024-03-09:
@hasecilu 
you said it very nicely.

shaise, 2024-03-10:
ok ok ok ok. 🤓
Lets do it. What needs to be done here, and sheetmetal as well?

kaktusus, 2024-03-10:
Of course.
Sheetmetal also.
These are very good working workshops. Congratulations on a successful "venture".  :smile: 

So I can upload the translation files in Crowdin?

shaise, 2024-03-10:
yes indeed.

hasecilu, 2024-07-01:
I think this could be closed now.

closed 2025-02-10 by luzpaz