shaise/FreeCAD_FastenersWB #349: ASME 18.2.1.6 UNC Hex Head Screw no longer visible

ppemawm, 2024-03-02:
This fastener is no longer visible in the Fasteners or Assembly4 workbenches.  I randomly checked a few others and did not find any other that were not visible but did not check all of them.  This occurred sometime in the last few up dates but I do not know exactly which.
[image]

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git)
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.7
  * DynamicData 2.60.0
  * fasteners 0.5.12
  * freecad.gears 1.2.0
  * MnesarcoUtils 0.2.5
  * QuickMeasure 2022.10.28
```

closed 2024-03-03 by shaise

Yrii77, 2024-03-04:
как скачать менеджер дополнений?

ppemawm, 2024-03-04:
> the fastener should be back now, please check

It is.  BTW I had to uninstall Fasteners with the AddOn workbench and re-install it in order for the changes to take effect.  Update is not enough even with restart.  

In any event, thank you, especially for the quick response.

PR #350 (2024-03-03): Make ASMEB18.2.1.6 reappear in 3D view
This screw was not included in the main control flow code making to raise an `NotImplementedError()`.

Fix #349
commit b3c1a809 (2024-03-03): Make ASMEB18.2.1.6 reappear in 3D view

This screw was not included in the main control flow code
making to raise an NotImplementedError().

Fix #349