shaise/FreeCAD_FastenersWB #351: Does fastners work with Links?

leoheck, 2024-03-06:
I believe this does not work yet. At least for me, using the latest FreeCad (0.22), if I select a reference in linked objects (I am using the new ~ondsel~ Assembly workbench)

What do you think if the fasteners workbench could support links? I believe the origin should be moved to the link origin to make this happen, right?

tobiasfalk, 2024-03-06:
See #307

leoheck, 2024-03-06:
@tobiasfalk did you link the right issue?

tobiasfalk, 2024-03-06:
Yes, because it Talks about including the Fasterner WB in to FreeCAD core to make a solide connection to the other Core WB's

leoheck, 2024-03-06:
Ah, that would be awesome, indeed. Thanks, O  going tô check that.

closed 2024-03-06 by leoheck

tobiasfalk, 2024-03-06:
I think closinh this may be a bit early, because such a Core implementation could and will take a bit of time, so if you could explain what exactly is not working, a "hot/temporary" fix could be done. If it is something that has to do with the Fasterner WB and not with the Assembly WB.

leoheck, 2024-03-06:
Thanks. I think this issue is related to the usage of Links with Fastners in general. The Assembly and other Assembly WBs use the links to add Parts without replicating them. 

I am attaching an example here. 
[assembly_link_test.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/14516911/assembly_link_test.zip)

I selected the hole of a linked part inside the Assembly and added on Fastener.
Normally, the screw adjusts itself to the hole, following its size and the placement centered in the center of the hole. 
However, in this case, the object is not a Part Design object but a Link to a Part Design object that is in a different place.
The fastener is being added to the origin of the linked objects instead of the origin of the link itself.
If you unhide the Base object, you are going to see where the screw was added.

[image]

leoheck, 2024-03-07:
The original Part `Base`, has the origin on 0,0,0, but the link was translated to 9,9,9

To make this happen I think it is a matter of detecting the object as a link, and then placing the screw translating the origin to where the link is... it also has to translate the rotation but I am not sure how hard is to do that.

[image]