shaise/FreeCAD_FastenersWB #352: Handling of Standards that define Bolt and Nut

tobiasfalk, 2024-03-06:
There are some Standards that define Bolts and Nuts, which meant to be used together. 
I wanted to ask if this should be implemented separate with a Bolt and a Nut, or maybe if a grouped approached should be taken?
An example for these are [EN 14399-4](https://www.beuth.de/en/standard/din-en-14399-4/207944502) and [EN 14399-8](https://www.beuth.de/en/standard/din-en-14399-8/274692933).

A grouped approach could look like a grouped creation and maybe even a grouped deletion (if one of the pair is deleted, the other one will also be)

shaise, 2024-03-10:
Interesting idea. But it has to be done without complicating the interface.

tobiasfalk, 2024-03-10:
One could just do one button to craze both Nut and Bolt and a check that when one is deleted the other one will also be.
An other, approach could be to to use the in New Assembly WB to groupe Nut and Bolt, but that would Limit the WB to only the newer FreeCAD versions.