shaise/FreeCAD_FastenersWB #355: "Add X Fastener" commands redundant

hasecilu, 2024-03-12:
Currently the description appears 2 times and the norm 3 times.

I think it's better to remove the description from the `Menutext` also I'm not sure if include again the norm in the `ToolTip` and don't know if the best place for the verb "Add" is the `MenuText` or the `ToolTip`.

Finally, the commands from main toolbar have "Fasteners_" prefix, the fasteners commands have "FS" maybe update that also

[image]

shaise, 2024-03-12:
I dont think we can change the FS<some-fastener> command. I think it will break loading of old files with fasteners.

hasecilu, 2024-03-12:
Didn't think about that, apart from that there is no other reason to change the command because there are no Wiki pages for that.

Only is left to discuss about MenuText and ToolTip. It seems that only the ToolTip has a way to shrink long string but MenuText not. I was thinking on having it more compact (visually).

[image]

It seems even for Russian special cases for singular/plural is used.

shaise, 2024-03-13:
yes, definitely we need to remove the redundancy. should look like your second image. But it should not affect the pull down mode:

[image]