shaise/FreeCAD_FastenersWB #360: Update icons

maxwxyz, 2024-04-13:
I want to adjust some icons, not a complete overhaul to match stroke size and appearance of the modification tools of Fasteners WB (like BOM, move, flip,...).
Currently they are blue, the WB icon is green.
@shaise should I make the modification tool icons green like the WB icon or should they stay blue (or yellow like parts) and the WB icon also changed to blue (or yellow)?
[image]

shaise, 2024-04-13:
Hi @maxwxyz 
I'm really open to any suggestion. right now the tools are blue, the Imperial fasteners are Cyan, and the metrics are orange.
Maybe we should pick 3 new colors from freecad's standard palette?

maxwxyz, 2024-04-13:
Following the discussion in the DWG I changed the modification icons to blue including the WB icon.

PR #361 (2024-04-13): Update modification icons to be consistent with FreeCAD
fixes #360 
Following the discussion in the DWG I changed the modification icons to blue including the WB icon.
Color palette of imperial vs metric and maybe a general overhaul could follow later.
If merged, I'm also in favor of https://github.com/shaise/FreeCAD_FastenersWB/issues/307

[image]
closed 2024-04-13 by shaise