shaise/FreeCAD_FastenersWB #362: UI: Calculator dropdown cut offd

maxwxyz, 2024-04-13:
Not sure if related to @obelist79 https://github.com/obelisk79/OpenDark/ theme, but the list in the dropdown of the screw calculator are cut off for most entries.
Same is also for the addon manager: https://github.com/FreeCAD/FreeCAD/pull/12859 see comments there.
It seems to work only with the classic theme.

[image]

shaise, 2024-04-13:
It looks like it uses a different font, and the aouto resizing of the drop down is not working correctly. I do not want to set an arbitrary minimum size in pixels, since it will not work nicely in different resolutions.

hasecilu, 2024-04-13:
I think is some OpenDark thing because when changing to Dark behave the entries are readable.