shaise/FreeCAD_FastenersWB #364: DIN ISO 14581

Roundwolf, 2024-04-17:
Hi all and @shaise , 

I managed to implement the DIN ISO 14581 into my FastenersWB. I would like to share the changes, so they can be implemented in the next Update. Where would be the best place to upload my files? Sorry for this question but I am very new to Coding generally. But my code works, except for the translation to other languages than English.

shaise, 2024-04-17:
Hi. 
Are you familiar with github and git in general?
Basically you need to fork this repository into your github account: 
[image]
Then clone it into your computer.
Do the changes then push them back to github.
Once you did it you need to create a "pull request", and this will automatically alert me that you have changes you want to merge. I can then do it automatically.

Roundwolf, 2024-04-17:
To be honest, I'm not familiar with git yet, so thanks for the explanation. I will try it the way you explained it. And I'll let you know if something doesn't work.

shaise, 2024-04-17:
Great!
Starting with git is not easy, but there are lots of video tutorials at youtube I'm sure.

closed 2024-04-17 by Roundwolf