shaise/FreeCAD_FastenersWB #367: [feature request] : adding standard injection molds  parts

mwlandolsi, 2024-04-25:
Is it possible to add some other types of parts that are used like fasteners in injection mold or die in sheet metal tools, such as ejecting pins, guiding rods, rings ...
I can provide dimensions and some standards.

mwlandolsi, 2024-04-26:
I think it is possible to add ejectors, sprue bushings, guiding elements ... more elements can be found here as an example : 
https://www.hasco.com/en/Product-catalogue/Z/Guide-elements/c/PIM0301
https://www.hasco.com/en/Produktkatalog/Z/Auswerfen/c/PIM0303
This supplier uses the most standard parts (ISO, DIN ...)

tobiasfalk, 2024-04-27:
1. I would not say it's not Possible, since we already have things that are not fasteners, it would just expand this to definitely not fasteners. @shaise?
2. The dimensions this supplier give are not complete, for example the dimensions of some groves are missing, and I was not able to find them somewhere else, in a quick search. This will make a possible implementation harder, so if you know of more detailed dimension charts for these parts, that would help the person who implements it.
3. Since the problem with the dimensions exist and the Fact that the interface is already pretty clustered(#363) but also since the supplier you linked to gives the capability to download a 3D model, I would not say it is a priority.

shaise, 2024-04-27:
I didn't see any ISO/DIN standards. I think we should concentrate on those for the time being.
As to @tobiasfalk 's items 2 and 3  I totaly agree

tobiasfalk, 2024-04-27:
@shaise the Norms are a bit hidden in the catalog, for example [ISO 8020](https://www.austrian-standards.at/en/shop/din-iso-8020-2003-07~p3074588) and [ISO8404](https://www.dinmedia.de/en/standard/din-iso-8404/223325003).
@mwlandolsi I would not close it since this would most like lead to it taking longer till it implemented.

mwlandolsi, 2024-04-27:
Maybe another similar workbench could be created to cover mold standard parts like ejector pins or bushings... They could be easily implemented like fasteners in tools rejects (molds and dies). In other hand, the interface is clustered and Fasteners WB is used for most common projects.

alexneufeld, 2024-04-27:
> Maybe another similar workbench could be created ...

You probably want to look into adding these sort of parts to https://github.com/FreeCAD/FreeCAD-library . The structure of that project is better equipped to handle new/disparate types of industrial parts.

hasecilu, 2024-04-27:
> You probably want to look into adding these sort of parts to https://github.com/FreeCAD/FreeCAD-library .

Good, idea!, I have submitted a lot of parts there, a temporary option would be to use a configuration table on a Spreadsheet ( I left some examples there). Having that would be make easier to later create the parts via code.

I also have been thinking on a general-purpose WB to submit "generic" parts to avoid creating `n` specific external WB, probably as an extension of the library. Also I have been waiting for material support (haven't checked lately) and built-in assembly capabilities to be able to make multi-part models as parametric Python objects.

mwlandolsi, 2024-04-27:
Freecad library is a good workbench but in this case and after several years of using freecad and fasteners wb, in addition to my experience in mold design, I see that fasteners wb is the ideal to use for mold parts implementation. The standard mold parts behave exactly like fasteners.

tobiasfalk, 2024-04-28:
I agree with @mwlandolsi the structure that I saw, in the quick look on the things I was able to find (mainly the links provided), is very very similar to the DIN and ISO Fasteners, if not identical (What I can not say since I did not see the standards themselves).
As already said in **2.** earlier, for any kind of implementation more detailed information is needed, since the links to the supplier give not detailed enough information on the dimensions, at least for the purposes of this Add-on.

@shaise How hard is it to add a WB that uses all the bases of the Fastener WB but only has Buttons for other stuff. This could also solve the cluster Problem with Nuts, since a Split between a Nut WB, a Screw and Bolt WB, a Nails WB and so on, could be a solution. But I must say a single Button solution where one has a Tree View to select and Idelay some kind of Parametric search is the best way.

shaise, 2024-04-28:
@tobiasfalk I think the best solution is a tree of fasteners PLUS the option to mark favorite ones to show on the toolbar for fast access. I'm definitely against multiple workbenches. For the time being I think I will just split the nuts to several groups.