shaise/FreeCAD_FastenersWB #371: SyntaxWarning: invalid escape sequence '\d'

sdifru, 2024-05-01:
Freecad shows this error `/home/baskara/.local/share/FreeCAD/Mod/fasteners/./FastenersCmd.py:613: SyntaxWarning: invalid escape sequence '\d'
  if len(re.findall("[.]\d*$", val)) > 0:`

Here's my freecad version:
```
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Installed mods: 
  * fasteners 0.5.20
```

shaise, 2024-05-01:
please explain how to reproduce this error step by step.

sdifru, 2024-05-01:
Its' gone now. I opened freecad and the errors showed in `Report view` panel. It showed right after I installed the plugin and restarted freecad. It seems like the problem is solved.

closed 2024-05-01 by sdifru