shaise/FreeCAD_FastenersWB #372: Not very happy with the shape of “GN 506 T-Slot nuts to swivel in”

berberic2, 2024-05-01:
With this shape it would not be possible to swivel the nut in. 

Image
Left: FreeCAD
Right: Item 0037001 Nutenstein 5 St M5
[image]

hasecilu, 2024-05-01:
Yeah, it need a tweak on the Bezier curve to be less round, the nut is from Ganter: https://www.elesa-ganter.in/en/ind/Machine-elements--T-Nuts--GN506#listtype=search&term=GN%20506

hasecilu, 2024-05-02:
Modifying the intermediate point(s) of the spline (blue) we can change the geometry.

Moving the point from the corner will help, maybe we need 2 points instead to make it less pointy.

Could you try to tweak the point(s) position(s) to make it similar to your other model?
[image]
[image]

berberic2, 2024-05-09:
Proposal:

[image]

PR: https://github.com/shaise/FreeCAD_FastenersWB/pull/378

closed 2024-05-15 by berberic2