shaise/FreeCAD_FastenersWB #377: Question: How to handle screws that are not available in all sizes in a range

berberic2, 2024-05-03:
P.e. „screw“ M20 is available in 80, 125 and 200 mm length but not in 100 and 160 mm. Of course the menu offers 100 and 160 mm as well, but these can not be generated because the length of the threaded part is only defined for known sizes…

For an example see https://github.com/berberic2/FreeCAD_FastenersWB/tree/din787

shaise, 2024-05-07:
If there is any vendor that offers these sizes, I have no issue with adding these extra sizes to the "known" sizes. As long as all the standardized fasteners are rendered correctly, I don's see any issue with some non standard extras.

tobiasfalk, 2024-05-07:
Could they be labeled with "NoS"(None Standard) so that one can directly see that they are not defined in the Standard.

shaise, 2024-07-23:
This is an interesting idea, Though I do not know if it worth the tedious job of going one by one over all possible options and update it.
Perhaps a small change to your idea - instead of having to implement the list for all diameters, we can do it only for the diameters that matter - so only if rangelist exists AND the selected diameter exists in the range list, only then take it. This way if we have a fastener with 40 diameters and only one of them is non-continues in range, we need to add only this one to the rangelist.