shaise/FreeCAD_FastenersWB #379: Thread crest missing after boolean cut (Part_Cut) a M30x20-Screw head with a cube (Part_Box)

lmutt, 2024-05-11:
I'm trying to remove (cut) the head from a "ISO 4017 Hex head screw". To accomplish this, I do the following steps:

1) Using the Fasteners Workbench, I add the fastener, configure it to be M30, length custom (20 mm), and select "enable" from "thread" property. All ok, the fastener and its thread looks good -- see [image].

2) Then I create a cube (Part_Box) with Part Workbench, resize and position it to make it cover the entire head of the screw (the part I want to eliminate) -- see [image].

3) Then I select first the screw object, second the cube and click the cut icon (Part_Cut). The screw is cut in the correct location, but the thread crest faces are now gone -- see [image]. I'm sure that the cut object is really missing "pieces", because when I export it in STL and import it in OrcaSlicer, I see the same thing: I can see the other side through the slits. It is no longer a solid object. OrcaSlicer complains about this, reporting error 1087 (non-manifold edges).

I'm doing something wrong? Is this a Freecad bug? or a fasteners bug?

INFO: Freecad/fasteners information obtained from Help->About menu:

> OS: Arch Linux (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.37249 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
> Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: Portuguese/Brazil (pt_BR)
> Installed mods: 
>   * fasteners 0.5.20

tobiasfalk, 2024-05-11:
Could you give us a File and a picture similar to the first or third but with the green Box?

lmutt, 2024-05-11:
> Could you give us a File and a picture similar to the first or third but with the green Box?

For sure. I have updated my original message with the required information.

tobiasfalk, 2024-05-12:
I can see your Problem in your file, but I am just not able to reproduce it from scratch, seem to me that this may be a DEV build problem, my guess is the OCC version since this was changed to a newer one and made some Problems. 
I do not think that this is a pure Fasteners WB problem, @shaise what do you think?
@lmutt when the next Weekly Build and if the Problem still exists (from new Objects and cuts to prevent corrupted files) and report about it. 
I tried it with: 
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * DynamicData 2.60.0
  * fasteners 0.5.20
  * kicadStepUpMod 11.1.0
```
And with
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * DynamicData 2.60.0
  * fasteners 0.5.20
  * kicadStepUpMod 11.1.0
```

shaise, 2024-05-13:
Generating threads is notoriously problematic with OCC. Sometimes the generated fastener is not a perfect solid and this can cause issues when more solid manipulations are done. But this is not something that can be solved within Fasteners WB.

lmutt, 2024-05-13:
As you said at the beginning, this specific problem seems to be related to the OCC version. Additionally, I noticed a strange artifact after cutting. (Please see updated the original message with "EDIT 2" and "EDIT 3").

lmutt, 2024-05-13:
Sorry for more messages... but looking carefully at just the pure fastener, recently added, I found that artifact face, see:
[image]

hasecilu, 2024-05-13:
Could be related with #375 ?

tobiasfalk, 2024-05-14:
> But this is not something that can be solved within Fasteners WB.  

@lmutt could you open a FreeCAD Issue and also link in it to this one.

lmutt, 2024-05-16:
> > But this is not something that can be solved within Fasteners WB.
> 
> @lmutt could you open a FreeCAD Issue and also link in it to this one.

I did more careful testing and edited my original message (EDIT 4) with the details I found. I now believe that this can be easily reproduced by you.

I think that the artifact present in the screw ("stray face") directly generated by Fasteners is causing this bug, and in this case it would not be appropriate to report this bug in Freecad. Maybe it's better if I open another bug here reporting only the problem of the artifact present in the pure screw, and link it to this issue. What do you think?

shaise, 2024-05-16:
What are the steps to get this stray face? I can not reproduce it. (though the missing crest is easily reproduces - without stray faces to start with)

lmutt, 2024-05-16:
> What are the steps to get this stray face? I can not reproduce it. (though the missing crest is easily reproduced - without stray faces to start with)

1) Select Fasteners Workbench
2) Add ISO 4017 hex head bolt
3) Select the above created screw. In Data tab, change the diameter property to at least M10
4) Set thread property to "true"
5) Rotate the 3d view (clicking in the navigation cube, in its vertice Right/Rear/Bottom). The stray face is visible at the starting of the thread (near the head). For M10 screw, the face is approximately in the position (x,y,z) = (3 mm, 4 mm, -0.55 mm) -- see figure below for its position.

[image]

tobiasfalk, 2024-05-16:
Yes I did see the artifacts but for some reason did not report them or ignored the, I may have to.much focused on the problem you described.
I will follow the steps you mentioned when I have access to a PC.

shaise, 2024-05-19:
Still can't reproduce the stray face. However cutting still results with missing crest.

lmutt, 2024-05-20:
> Still can't reproduce the stray face. However cutting still results with missing crest.

Tested here again...  with the latest Freecad weekly build available, and this time with a clean Freecad config:

```
rm -rf ~/.cache/FreeCAD/
mv ~/.config/FreeCAD/ ~/.config/FreeCAD.bak
mv .local/share/FreeCAD/ .local/share/FreeCAD.bak
```

Then starting the "brand new" Freecad and downloading again the fasteners WB.

The stray face are reproduced again easily. Just adding an ISO 4017 hex head bolt and setting it for the diameter M10, enabling threads, and pressing 5 on keyboard to see the bottom side. Please see attached screenshot with the stray face highlighted. Does this problem (stray face) appears just for me?

[image]

Version info:

```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * fasteners 0.5.21
```