shaise/FreeCAD_FastenersWB #381: AutoDiameter fails if hole is too large

Roy-043, 2024-05-21:
1. Create a hole with a diameter of 100mm (users can do strange things... :wink:)
2. Select the circular edge.
3. Insert a fastener, for example DIN571.
4. Result: Error message see below.

```
21:53:37  pyException: Traceback (most recent call last):
  File "C:\Users\Name\AppData\Roaming\FreeCAD\Mod\fasteners\.\FastenersCmd.py", line 669, in execute
    self.calc_diam = screwMaker.AutoDiameter(
                     ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Name\AppData\Roaming\FreeCAD\Mod\fasteners\.\ScrewMaker.py", line 487, in AutoDiameter
    return res
           ^^^
<class 'UnboundLocalError'>: cannot access local variable 'res' where it is not associated with a value
21:53:37  Screw: cannot access local variable 'res' where it is not associated with a value
```

shaise, 2024-05-22:
Thanks! I'll take a look.

shaise, 2024-05-24:
Should be fixed now.

closed 2025-02-10 by luzpaz