shaise/FreeCAD_FastenersWB #385: Failed to read package.xml

FEA-eng, 2024-05-31:
I have Fasteners installed and updated and each time I open the Preferences (without even going to this workbench first), I see this:

`Failed to read C:\Users\username\AppData\Roaming\FreeCAD\Mod\fasteners.backup1702501118.9117603\package.xml`

Can this be solved on your side? The workbench itself seems to be working fine for me.

```
OS: Windows 10 build 19041
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

shaise, 2024-05-31:
try refreshing the cache of the addon manager
[image]

FEA-eng, 2024-06-03:
> try refreshing the cache of the addon manager

Thanks. The issue is gone, at least for now. On one computer I just uninstalled Fasteners, deleted those files and reinstalled and it's working. On other computers, I don't see this error anymore. But if it comes back in the next weekly builds for some reason then I'll follow your advice.

hasecilu, 2025-01-06:
This should be closed now?