shaise/FreeCAD_FastenersWB #387: Undo crashes FreeCAD.

czinehuba, 2024-06-04:
Create a new body 
Create a sketch of a rectangle
Create a base wall
Create a bend on one of the edges
Press Ctrl+z to undo.
FreeCAD closes straight away....

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-wayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git) Snap 1040
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.5.21
  * sheetmetal 0.4.16
  * lattice2 1.0.0
  * Nodes 0.1.36
  * 3DfindIT 1.2.0
  * freecad_metal_workbench 0.0.1
  * Defeaturing 1.2.2
  * CfdOF.backup1708442124.622556 1.21.7 (Disabled)
  * CfdOF 1.25.17
  * OpenTheme 2024.5.16
  * freecad.gears 1.2.0

hasecilu, 2024-06-04:
I tested on latest weekly build and it does not close FreeCAD. Also should this be on https://github.com/shaise/FreeCAD_SheetMetal instead?

czinehuba, 2024-06-05:
Yes. It should have been in sheetmetal. Sorry. I'll close this and do further testing on my other PC.

closed 2024-06-05 by czinehuba