shaise/FreeCAD_FastenersWB #388: Unsupported operand type(s) for +: 'PySide2.QtWidgets.QDialogButtonBox.StandardButton' and 'PySide2.QtWidgets.QDialogButtonBox.StandardButton'

FloatingPoint75, 2024-06-08:
Hello,
when I launch the "Change Parameters" command I get the following error:

17:23:56  Traceback (most recent call last):
  File "C:\Users\giuli\AppData\Roaming\FreeCAD\Mod\fasteners\.\FSChangeParams.py", line 403, in getStandardButtons
    return QtGui.QDialogButtonBox.Ok + QtGui.QDialogButtonBox.Cancel
<class 'TypeError'>: unsupported operand type(s) for +: 'PySide2.QtWidgets.QDialogButtonBox.StandardButton' and 'PySide2.QtWidgets.QDialogButtonBox.StandardButton'

The command works anyway, but it looks like there is a wrong operand (That should be a | and not a +)

Thank you!

shaise, 2024-06-08:
Thanks for the catch!
Should be fixed now.

FloatingPoint75, 2024-06-08:
Great, thanks for the quick fix!