shaise/FreeCAD_FastenersWB #390: 'ThreadedRodlength': Mapping key not found.

markkco, 2024-06-16:
Error message when trying to change threaded rod length via `Change fastener parameters`.

Message:
```
xxx\Roaming\FreeCAD\Mod\fasteners\.\FSChangeParams.py Line 384
xxx\Roaming\FreeCAD\Mod\fasteners\.\ScrewMaker.py Line 396
xxx\Roaming\FreeCAD\Mod\fasteners\.\ScrewMaker.py Line 554
'ThreadedRodlength': Mapping key not found.
```

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Croatian/Croatia (hr_HR)
Installed mods: 
  * A2plus 0.4.65
  * Assembly4 0.50.13
  * BIM 2021.12.0
  * EM
  * fasteners 0.5.22
  * NikraDAP 2.0.0
  * ProDarkThemePreferencePack 1.0.0

shaise, 2024-06-17:
This is strange. `Change fastener parameters` does not support threaded rods at all. It should have been grayed out:
[image]

shaise, 2024-06-17:
bug confirmed. checking.

shaise, 2024-06-17:
bug fixed. please update