shaise/FreeCAD_FastenersWB #391: [Improvement] Properties are displayed in lower case

leoheck, 2024-06-18:
Hi, can we have the properties using Sentence case or Title Case as it is with the Base properties?
Values false/true could also be False/True.
[image]

shaise, 2024-06-18:
the true/false are built in FreeCad properties. they are not controlled by fasteners workbench

leoheck, 2024-06-18:
I see, but this list is a display that does not need to be the Python code itself. You can treat them to display to the user. You can also treat them to get from the user. Freecad is getting polished on the dev version and I thought the text on this extension could be improved too.

shaise, 2024-06-18:
Perhaps I was not clear. I can not change it even if I want to. This is an internal FreeCAD featute. 
For example: sketch parameters are also lower case:
[image]

leoheck, 2024-06-18:
Ah, I see. Then if it is like that everywhere, it is good as it is. Now property names could still be improved to match the other properties, right?

shaise, 2024-06-18:
Yes! Regarding the name, I believe its possible

shaise, 2024-06-18:
And... Sorry. Although this name is controlled by the workbench, this is also saved when saving the file. If I change everything to upper case, it will no longer be able to read old files...

leoheck, 2024-06-18:
I see, why don't you show this using Title Case and when saving you do in lowercase... or even better ignore the case when reading from the file so old versions will be ok, and the new versions of the file will be written in title case.

leoheck, 2024-06-18:
I think you can a simple regexp do filter out the case when reading from the file.

shaise, 2024-06-18:
> I see, why don't you present them using Title Case, and when saving you do it in lowercase... or even better ignore the case when reading from the file so old versions will be ok, and the new versions of the file will be written in title case.

Once again - this is internal to freecad. That's why you can not translate property names.
The only way to fix it currently, Is create a conversion script that read old files, copy old parameters to new parameters and delete old parameters. Also, every place there is a use for the parametes (fastener WB files or any script that uses it) must be converted as well. I wish that there was a possibility to have a separate display name - this way it was possible to change only the display name and even have translations for it.
Perhaps Freecad will eventually support it.

leoheck, 2024-06-18:
Thanks for the clarification @shaise

leoheck, 2024-06-19:
Oh, that's beautiful. Nice @shaise. Thanks for considering this suggestion. As soon as I install it I will test my old files to report in case of any problem.

leoheck, 2024-06-19:
I am checking this right now, this looks good and professional.

[image]

Did not have any issues with a couple of files.
I will close this issue, and if I have any I can open it again, or create a new one. Cool?

closed 2024-06-19 by leoheck