shaise/FreeCAD_FastenersWB #393: Changine size of fastener results in error

Sil68, 2024-06-26:
When trying to change the size of a fastener ('change fastener parameter', 'set size') I am always receiving an error message:

```
At /Users/user/Library/Application Support/FreeCAD/Mod/fasteners/./FSChangeParams.py Line 384
08:06:06  At /Users/user/Library/Application Support/FreeCAD/Mod/fasteners/./ScrewMaker.py Line 397
08:06:06  At /Users/user/Library/Application Support/FreeCAD/Mod/fasteners/./FastenerBase.py Line 280
08:06:06  argument of type 'float' is not iterable: Inappropriate argument type.
```

I have tried this both on Debian Linux as well as on MacOS, rendering the very same results.

```
OS: macOS 14.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.23
  * ThreadProfile 1.90.0
  * Reporting
  * btl 0.9.9
  * BillOfMaterials 0.0.8.5
  * freecad.gears 1.2.0
  * woodworking 0.21.2.33771
  * OpenTheme 2024.5.16
  * Defeaturing 1.2.2
  * Gridfinity 0.5.0
  * FEMbyGEN 2.1.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.23
  * lattice2 1.0.0
  * Estimate 0.1.2
  * Design456 0.0.1
  * WB_Organizer 2024.1.29
  * sheetmetal 0.4.18
  * FEM_FrontISTR 0.1.0
  * 3D_Printing_Tools
  * Curves 0.6.39
  * Assembly4 0.50.13
```

valpackett, 2024-08-18:
Just hit this as well. As a quick fix, FSChangeParams.py Line 384:

```python
                                d, l, w = screwMaker.FindClosest(
```

and in ScrewMaker.py

```python
        # auto find width, if applicable
        if width is not None:
            width = str(width) # <<<<<<< add this
            width_table = FsData[type + "width"][diam]
```

```python
        # auto find length
        len = str(len) # <<<<<<< add this
        lens = self.GetAllLengths(type, diam, False, width)
        origlen = FastenerBase.LenStr2Num(len)
```

shaise, 2024-08-31:
@valpackett thanks!
@Sil68 , it should be fixed now.

closed 2025-02-10 by luzpaz