shaise/FreeCAD_FastenersWB #394: Add angle shift / offset for fasteners

maxwxyz, 2024-07-21:
From https://forum.freecad.org/viewtopic.php?p=614456#p614456

It would be great to have also an angle offset (like the length offset) to shift the angle from the default placement.
If you want the nut to be linked to the base object but want to edit the rotation in the placement dialog or via the transform tool it always resets after applying the transform. 
There is currently no solution if the part has a groove for the nut and you need to rotate the nut 30° around its thread axis.

[image]

@shaise FYI, I've created the issue here from the forum thread to not loose track.

alexneufeld, 2024-07-21:
A more comprehensive solution would be to switch to using Part::AttachExtensionPython 
https://wiki.freecad.org/Scripted_objects_with_attachment
This would also improve support for other common cases of fastener placement, such as slots or square holes

shaise, 2024-07-22:
I will add this option, but as I understand with the integration of assembly wb, the right way to do it is using the assembly?

maxwxyz, 2024-07-22:
Not sure if therefore an Update in Assembly WB is necessary? If so, I think this won't be done for 1.0 I guess.

shaise, 2024-07-22:
I think assembly supports orientation of joints.

shaise, 2024-07-25:
Option to rotate the fastener added.
[image]

maxwxyz, 2024-07-25:
Thanks, I'll close it then.

closed 2024-07-25 by maxwxyz