shaise/FreeCAD_FastenersWB #397: Fastener name of Spacer and Standoff with double x

maxe22, 2024-08-16:
The auto generated name at fasteners Spacers and Standoff have two "x" between the length and width.
[image]

Is that desired?
In my opinion there should be only one "x", like
[image]

In FastenersCmd.py the value for "dispWidth = 'x' + fp.Width" should be changed to "dispWidth = fp.Width"
```
      # Formation of fastener name: DxLxH(LH)-Type
        dispDiam = self.CleanDecimals(self.calc_diam)
        label = dispDiam
        if hasattr(fp, 'Length'):
            dispLen = self.ActiveLength(fp)
            label += 'x' + dispLen
            if hasattr(fp, 'Width'):
                dispWidth = fp.Width
                label += 'x' + dispWidth
```

shaise, 2024-08-17:
You are right! fixed. Thanks for the catch.

luzpaz, 2024-08-17:
CLose ?

shaise, 2024-08-17:
I usually wait for the issuer to close.

maxe22, 2024-08-17:
I have tested it and it can be closed. Thanks!

closed 2024-08-17 by maxe22