shaise/FreeCAD_FastenersWB #398: Numerous error messages, no fasteners displayed

StSav012, 2024-08-26:
The file attached (remove the “.zip” suffix once you get the file, do not unpack) produces countless errors in the console. Some of them look like typos, some are harder to resolve. Anyway, the model misses all the fasteners.

First, this appears:
```
Traceback (most recent call last):
  File "/home/stsav012/.local/share/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 615, in onDocumentRestored
    self.VerifyMissingAttrs(obj)
  File "/home/stsav012/.local/share/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 474, in VerifyMissingAttrs
    self.updateProps(obj)
  File "/home/stsav012/.local/share/FreeCAD/Mod/fasteners/./FastenerBase.py", line 89, in updateProps
    linkedObj = obj.BaseObject
                ^^^^^^^^^^^^^^
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'BaseObject'
```
There should be `baseObject` instead of `BaseObject`.

Then it comes
```
Traceback (most recent call last):
  File "/home/stsav012/.local/share/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 948, in onDocumentRestored
    self.originalType = obj.Proxy.Type
                        ^^^^^^^^^^^^^^
<class 'AttributeError'>: 'FSThreadedRodObject' object has no attribute 'Type'
```
And so on.

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.27
```

[plate.FCStd](https://github.com/user-attachments/files/16750769/plate.FCStd.zip) (remove the “.zip” suffix, do not unpack)

luzpaz, 2024-08-26:
@StSav012 would you like to submit a PR ?

StSav012, 2024-08-26:
Unfortunately, I have no idea how to resolve most of the errors. I've spent about an hour, but I couldn't make it work. The error I've given up was just a “syntax error”, at a seemingly valid code:
```python
            if addCustomLen:
                obj.addProperty("App::PropertyLength", "LengthCustom", "Parameters", translate(
                    "FastenerCmd", "Custom length")).LengthCustom = self.inswap(slens[0])
```
Maybe, it's a result of my erroneous corrections or a FreeCAD bug after all. IDK. I don't get the logic behind most of the code.

And yeah, the code style could be better, without all the `builtins` overrides, with docstrings or typing annotations, at least. I've shot my leg trying to make sense of the objects here and there.

closed 2024-08-26 by shaise

StSav012, 2024-08-27:
Thank you so much! There no errors appear now.

PR #399 (2024-08-26): Fix attempt of reading unexistent parameters
Fix #398
commit 500c9a49 (2024-08-26): Fix attempt of reading unexistent parameters

Fix #398