shaise/FreeCAD_FastenersWB #403: Fasteners Workbench not loaded

Norbert-Rotorwerk, 2024-10-14:
The fasteners workbench is not loaded and is not available within the menu tools --> addon-manager within the 1.0RC2 and the weekly-build releases.

Norbert

OS: Debian GNU/Linux 12 (bookworm) (MATE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.4.55
  * BOLTSFC 2022.11.5
  * parts_library.bak
  * parts_library
  * symbols_library
  * Defeaturing 1.2.0
  * 3DfindIT 1.2.0
  * Assembly4 0.12.6
  * Manipulator 1.5.0
  * MeshRemodel 1.8919.0
  * Render 2023.2.2
  * sheetmetal 0.2.61
  * Help 1.0.3
  * FEM_FrontISTR 0.1.0
  * Assembly3 0.12.2

shaise, 2024-10-14:
please look at the report view after loading freecad and see if you have something like the following line:
```
18:02:31  Init:      Initializing C:\Users\shai\AppData\Roaming\FreeCAD\Mod\fasteners\.\... done
```
see if there are errors near that line.
I have no control over the addon manager if indeed it a manager bug.

luzpaz, 2024-10-14:
> fasteners 0.4.55

Wait.. latest version fasteners is v0.5.29

shaise, 2024-10-14:
@luzpaz , Again its the issue of addon manager not detecting or unable to update the version. Many times I suggested users to uninstall/install the the addon in order to get the recent version.

shaise, 2024-10-14:
@Norbert-Rotorwerk , can you try to uninstall/install the workbench? if you don't see it in the addon manager check if the "filter" drop down at the top is set to any, any.

luzpaz, 2024-10-14:
> Again its the issue of addon manager not detecting or unable to update the version. Many times I suggested users to uninstall/install the the addon in order to get the recent version.

@shaise i'm blurry on the details of this. Did you mention this to me once? is there a ticket open for it ?

Norbert-Rotorwerk, 2024-10-15:
I uninstalled and installed the fastener workbench. After that the workbench was loaded without any issue. From my point of view it could be an issue with the addon-manager.

shaise, 2024-10-15:
@luzpaz ,
I dont know if there is a ticket. I get this issues on both sheetmetal and fasteners. However it seems no other WB report this problem, so perhaps I'm doing something wrong. I do not know if it was you or another admin I told it to, but the response was that my package.xml is correct, so I really dont know what to do about it.

luzpaz, 2024-10-15:
@chennes have you seen/heard of this bug before ? https://github.com/shaise/FreeCAD_FastenersWB/issues/403#issuecomment-2411855137