shaise/FreeCAD_FastenersWB #410: DefaultFastenerColor is 100 % transparent.

berberic2, 2024-11-10:
If _DefaultFastenerColor_ is set in the preferences, the created screw has the property _Transparency_ set to 100 %. Only the lines are visible.

The problem is that `FSParam.GetUnsigned("DefaultFastenerColor", 0xcccccc00)` returns a value with the lowest 8 bits set to 1 (0x??????FF), so having an alpha Value of 0xFF. It is not possible to set the alpha-value in the color-selection-dialog.

workaround:
```
diff --git a/FastenersCmd.py b/FastenersCmd.py
index 4e49727..8b7c4ba 100644
--- a/FastenersCmd.py
+++ b/FastenersCmd.py
@@ -885,7 +885,7 @@ class FSScrewCommand:
             FSScrewObject(a, self.Type, selObj)
             a.Label = a.Proxy.familyType
             if FSParam.GetBool("DefaultFastenerColorActive", False):
-                a.ViewObject.DiffuseColor = FSParam.GetUnsigned("DefaultFastenerColor", 0xcccccc00)
+                a.ViewObject.DiffuseColor = FSParam.GetUnsigned("DefaultFastenerColor", 0xcccccc00) & 0xFFFFFF00
             if FSParam.GetBool("DefaultLineWidthActive", False):
                 a.ViewObject.LineWidth = FSParam.GetFloat("DefaultLineWidth", 1.0)
             if FSParam.GetBool("DefaultVertexSizeActive", False):

```

shaise, 2024-11-11:
Wow this is so strange. An alpha value of 0xFF should be fully opaque (in all standards that I know of) 
why is it opposite in freecad?

shaise, 2024-11-11:
I KNOW it used to work before. So something must have changed. Anyway at least your change seems to be backward compatible

maxwxyz, 2025-02-04:
is this issue still relevant with the changes in main?