shaise/FreeCAD_FastenersWB #414: APPLY doesn't work in PLACEMENT

fester225, 2024-12-05:
OS: macOS 15.1.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * AnimationFreeCAD 1.0.0
  * pcb 6.2023.1
  * offline-documentation 1.0.0-alpha
  * kicadStepUpMod 11.4.4
  * fasteners 0.5.33
  * AirPlaneDesign 0.4.1
  * Help 1.0.3

Thank you for this utility! Until now I've been spending much time downloading CAD drawings from McMaster-Carr.

I selected 7/16" x 3/4" Set Screw.
Clicked on PLACEMENT
Successfully moved the set screw to where I wanted it
Clicked on APPLY
The set screw went back to it's original position

This has happened 4 times. Can anything be done about this?

fester225, 2024-12-05:
I just got the same results using DATA/BASE from the main display. I can successfully move the set screw, but the new position is not retained (the set screw goes back to where it started).

fester225, 2024-12-05:
I am using FASTENERS workbench in a PART drawing.

hasecilu, 2024-12-05:
Can't reproduce on 1.0.0, could you make a small video

shaise, 2024-12-05:
Please attach a sample file

fester225, 2024-12-05:
Cancel This Problem.
I tried doing the same thing again, but this time FreeCAD is behaving the way it's supposed to. Last time I asked for an IMP flat point set screw I got 7/16"x1/2". This time I get a 1/4"x3/8" flat point set screw, and it behaves the way it's supposed to.

shaise, 2024-12-05:
Just FYI, if you click (select) any curved edge in your model, and then create a fastener, the fastener will automatically center to that curved line. If you try to change the fastener location, it will always snap back to the original position.

fester225, 2024-12-05:
> If you try to change the fastener location, it will always snap back to the original position.

Sounds like what I might have done. Why is this helpful?

shaise, 2024-12-05:
this lets you place fasteners in appropriate holes of your model, without the need to create and manually place the fastener. (you can even select multiple holes)
This way the fasteners also follow the model if change its location.