shaise/FreeCAD_FastenersWB #418: Translations missing even if they are translated

maxe22, 2025-01-05:
At the German translations I recoginzied some small spelling issues in translations. I corrected them at Cowdin. Also I translated all the other open expressions in German language. 
During that I saw that some expressions are already translated, but they are still showed in English. I changed the language to Spanish and the situation is the same. The expressions were translated serval month ago, and the last update with translations was made end of November. So it should be included.

That are the expressions which are translated in German and Spanish, but showing only in English:
FSScrewCalc.py
` 'MenuText': _translate("DockWidget", "Screw calculator", None),`
`'ToolTip': _translate("DockWidget", "Show a screw hole calculator", None)`
FastenersCmd.py
`'ToolTip': _translate("DlgChangeParams", "Change parameters of selected fasteners", None)`

At screw calculator will be currently the first combo for the fastener type not translated. In my opinion it make sense to translate also that.
FSScrewCalc.py
`FSCScrewTypes = (
    ("ISO7045.svg", "Metric Screw", ScrewMaker.FSCScrewHoleChart),
    ("PEMPressNut.svg", "PEM Press-nut", FSCPEMPressNutHoleChart),
    ("PEMBLStandoff.svg", "PEM Stand-off", FSCPEMStandOffHoleChart),
    ("PEMStud.svg", "PEM Stud", FSCPEMStudHoleChart),
    ("ASMEB18.2.1.6.svg", "Inch Screw", ScrewMaker.FSC_Inch_ScrewHoleChart)
)`

shaise, 2025-01-06:
The usage of crowdin is not maintained by me, so I need to wait until the maintainers update the WB with the new translations.

hasecilu, 2025-01-06:
I think strings are translated because of the `.ui` files but since the `.py` files use `_translate()` (with underscore) and that prevents `lupdate` to pick it.

PR #419 (2025-01-07): Include missing strings to translation
Hi @maxe22 did you detected other missing strings?

TODO:
- [ ] tweak code to allow bigger drop-down and prevent text shortened
[image]

Fix #418
closed 2025-01-07 by shaise