shaise/FreeCAD_FastenersWB #420: ScrewTap invalid shape after `refine=True`

Rapsssito, 2025-01-14:
I am getting `Boolean operation: Invalid curve on surface` with a boolean object with `refine=True` that contains a ScrewTap object with specific characteristics.

Steps to reproduce:
1. Add a ScrewTap with: custom diameter of 29 mm, 5 mm length and 1.5 mm pitch custom.
2. Move to Part Workbench.
3. Add a Cube.
4. Create a Fusion of the Cube and the ScrewTap.
5. Set the Fusion `refine` parameter to true.
6. Check the Part Geometry (with boolean operation check).

The output of the check is `Boolean operation: Invalid curve on surface` for several surfaces and edges of the screw.

With `refine=False`, there is no error (though the STL generated from the Fusion object is still "in bad shape"). If I make small changes to the length, the diameter or the pitch of the ScrewTap, the error no longer appears.

shaise, 2025-01-14:
Helix models in FreeCAD are notoriously problematic. Thus is a problem with the core OCCT library.

shaise, 2025-01-14:
by the way, why do you use screw tap and not threaded rod? the screw tap is the inverse of a screw and used to cut threads in a solid hole.

Rapsssito, 2025-01-14:
Thanks for the answer. I will close this issue then. I used a screw tap instead of the threaded rod because I wanted to avoid the chamfered corners. I will try with a threaded rod.

closed 2025-01-14 by Rapsssito