shaise/FreeCAD_FastenersWB #423: Freecad Crashes using Fastners

leoheck, 2025-01-31:
I don't know how I trigger this behavior, I am just sharing the command line output of the crash since it may be useful...

[image]

Text version
```
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 675, in execute
    self.calc_diam = screwMaker.AutoDiameter(
                     ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lheck/.local/share/FreeCAD/Mod/fasteners/./ScrewMaker.py", line 487, in AutoDiameter
    return res
           ^^^
<class 'UnboundLocalError'>: cannot access local variable 'res' where it is not associated with a value
Washer: cannot access local variable 'res' where it is not associated with a value
ControlSingleton::showDialog: Can't show  Gui::TaskView::TaskDialog  since there is already an active task dialog
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x732f5d445320]
#1  [0x6242cf78b4c0]
```

I am using Freecad 1.1-dev

luzpaz, 2025-01-31:
Full About info so we can also see the version of Fasteners you're using too

leoheck, 2025-01-31:
@luzpaz, updated above.

shaise, 2025-02-02:
Is this reproducible?

luzpaz, 2025-02-04:
No steps to reproduce but perhaps we should keep open for the error message ?

ysard, 2025-06-14:
Hi, I have this error too in a reproducible way. It happens when I have a pocket made in another pocket. When selecting the inner pocket edge and adding a screw like "ISO 7046", it generates the error. Changing the screw diameter from "Auto" to "M2.5" solves the issue. If I select the screw "ISO 7045" for example, the diameter is automatically selected on "M3" and there is no error.

In the uploaded file (see attachment), the edge selected is the following one:
    
```python
>>> Gui.Selection.addSelection('Unnamed','Body010','Pocket013.Edge9',0.629169,1.02183,5)
```

I hope this helps.

Error:
```
07:53:00  Traceback (most recent call last):
  File "/home/my_name/.local/share/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 675, in execute
    self.calc_diam = screwMaker.AutoDiameter(
                     ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/my_name/.local/share/FreeCAD/Mod/fasteners/./ScrewMaker.py", line 487, in AutoDiameter
    return res
           ^^^
<class 'UnboundLocalError'>: cannot access local variable 'res' where it is not associated with a value
07:53:00  2944.8 <App> FastenersCmd.py(894)|Document.cpp(3427): Failed to recompute Unnamed#Screw: cannot access local variable 'res' where it is not associated with a value
07:53:00  Recompute failed!
```

Installed version: `Fasteners Workbench 0.5.29`.

```
OS: xxx Linux Rolling (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * freecad.gears 1.2.0
  * FreeCAD-GDT 0.20.1
  * Render 2024.9.25
  * Defeaturing 1.2.2
  * AnimationFreeCAD 1.0.0
  * fasteners 0.5.29
```

Test File:
[screw_error.tar.gz](https://github.com/user-attachments/files/20735947/screw_error.tar.gz)

shaise, 2025-08-09:
@ysard , Thanks! 
Got it. This happens when Fasteners can not find any matching screw. I changed it so it will at least enter some default one.
should be fixed now.

shai

luzpaz, 2025-08-09:
Closing. Please re-open if necessary.

closed 2025-08-09 by luzpaz