shaise/FreeCAD_FastenersWB #426: [Feature Request] Size calculation for non threaded holes

mgth, 2025-02-10:
I could not find if the option exists, but fasteners size seams to be calculated with threaded holes in mind.
In my usage the part where the fastener is attached is usually a passthrough hole. And an oversized fastener is selected by default.

The best option would be to detect the hole threaded size property instead of relying on circle size.
If not doable, maybe a selectable option in the toolbar to switch between threaded/non threaded.

shaise, 2025-02-10:
The following tool buttons decides if an auto pass through fastener is selected or auto thread hole.

[image]

mgth, 2025-02-10:
Sorry, that was in front of me.

closed 2025-02-10 by mgth