shaise/FreeCAD_FastenersWB #428: Feature Request: Threaded inserts

leoheck, 2025-02-14:
Hi, is there any plan to add threaded inserts like this?

[image]

shaise, 2025-02-16:
we already have these:

[image]

Do you have any standard specification for yours?

leoheck, 2025-02-16:
Oh, great. I could not find it.
Honestly, there are some many items, it is hard to find, even if looking from the menu instead of the toolbar.

I don't have any specific specs for now, I just want to check if I can use the existing one first.

Off-topic: _The fastners wb is so useful it needs a better way for finding and selecting the fasteners.
It could have a widget that you type something and it lists items fasteners that match and display those things in a list with images that are not small. Also the input field should accept a regexp so users can find things easily._

shaise, 2025-02-16:
> It could have a widget that you type something and it lists items fasteners that match

Agree!

leoheck, 2025-02-16:
Thanks shaise, I am going to play a bit with the threaded inserts.

leoheck, 2025-02-16:
Ah, saw it is called `Heat Staked Metric Insert`, I read this but thought it was not what I was looking for. The image was too small I guess.

leoheck, 2025-02-16:
There is only a custom lenght, is this right?
[image]

closed 2025-02-16 by leoheck

shaise, 2025-02-16:
> There is only a custom lenght, is this right?

It looks like so

luzpaz, 2025-02-16:
Maybe the searchbar addon can be leveraged ?

leoheck, 2025-02-16:
I am going to create a new issue for that.