shaise/FreeCAD_FastenersWB #429: Feature Request: Widget to find and list fasteners

leoheck, 2025-02-16:
This is a simple idea to make the task of finding fasteners easier since this WB is more like a library and we have tons of toolbar buttons. 2 rows of buttons are fine, but Fasteners WB has so many.

It could be a simple input field to list items, that uses regexp.

Other options could be added to make the resulting list smaller, for instance, a checkbox to enable/disable Metric/Imperial fasteners.

shaise, 2025-02-16:
By the way, I always use the drop-down mode, otherwise its too much space.

[image]

leoheck, 2025-02-16:
Hm, thanks. Even tho, this mode looks a only bit better than the standard mode, honestly.
I prefer to find things using the top Menu then, because. the placement is organized, it is just a list of items.
The toolbars keep changing a lot, especially when Freecad crashes, things change and it is horrible to find things this way.
The drawback is that the Menu has small icons and I cannot filter out things that I don't want to look for.
A better solution for increased usability is s list where we can find things by typing substrings

leoheck, 2025-02-16:
https://github.com/shaise/FreeCAD_FastenersWB/issues/428#issuecomment-2661457147

It could be a toolbar button with a taskbar that has this.
It can be launched by default when opening the Fastenrers WB. 
And then mostly all those toolbar items could be gone.
Leaving only the important ones.

shaise, 2025-02-16:
I think the best option is a search field embedded in the tool bar, when you start typing it will open a window below with suggestions. But need to check how complicated it is

leoheck, 2025-02-16:
this would be a suboptimall implementation since I could let the task widget open with the resulting query to keep reusing things. for instance if my design uses 4 different types of fasteners, I could set the search field with something that gives me the exact types I need and only that. 

But I am not against such an Improvement, so if you think a toolbar search is good enough, let's see how it will work. 

But don't forget the most important part, having a shortcut to focus the search bar, and expand the current selection is user wants to use something that he found before and to have the whole search text selected in the case the user wants to change something else easily.

hasecilu, 2025-02-21:
An option could be to use the **Search Bar** addon forked by APEbbers (available on the Addon Manager). With that you can seearch for fastenres using the norm because norms are included on the descriptions (MenuText of the commands).

Here I searched "iso 74" and matching fasteners appear. Using also Ribbon UI from APEEbbers btw.

[image]