shaise/FreeCAD_FastenersWB #435: Need M1, M1.2, M1.4, and M1.7 options for metric screws.

Xpendable172, 2025-03-23:
I need 1mm, 1.2mm, 1.4mm, and 1.7mm options for metric screws (cross hatch types) for small intricate parts for the OpenHarrier cockpit simulator project. I'm working on some parts that need the 1.4mm and 1.7mm screws and unable to create those screws, which is hindering me.

shaise, 2025-03-30:
Hi, do you have any specs I can take dimension of?

Xpendable172, 2025-03-31:
I don't have a lot of information, but here's what I do have:

[image]

[image]

This is a box of screws that I purchased from this link:
https://www.amazon.com/Phillips-Micro-Screws-Self-Tapping-Electronic/dp/B07GKXMSVR/ref=asc_df_B07GKXMSVR?mcid=db313c6ceb973049b54ae0bd88d86162&hvocijid=9975288274183444881-B07GKXMSVR-&hvexpln=73&tag=hyprod-20&linkCode=df0&hvadid=721245378154&hvpos=&hvnetw=g&hvrand=9975288274183444881&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9021427&hvtargid=pla-2281435177178&th=1

I hope this helps!

shaise, 2025-03-31:
Hmmmm these are self tapping screws. Not sure it is easy to find a standard for.
I need to investigate further. Hope I find something.

Xpendable172, 2025-03-31:
That's okay.  They probably don't have to be self-tapping screws.  Something in the ball park would probably be sufficient, even if they were machine screws. I just need something to represent those screws.  In the meantime, I can fudge them by making my own simple model.

luzpaz, 2026-09-12:
Did we find the standards for them ?

shaise, 2026-09-12:
@luzpaz sorry, no standard found for them. need the official ID to search

g3grau, 2026-09-21:
I would also love to see M1 etc. for hexagon socket head screws/washers (or any other shape). I think M1.6 is currently the smallest one in the WB? No idea where to find valid reference data, but e.g. [machiningdoctor](https://www.machiningdoctor.com/threadinfo/?tid=1000) seems to have a lot of data. There's no easy way for a parametric custom screw, I guess?