shaise/FreeCAD_FastenersWB #437: ScrewTap length

tarman3, 2025-03-30:
Can not input some values of the length for ScrewTap
Problem only with _FreeCAD_weekly-builds-40971_

https://github.com/user-attachments/assets/66504f92-2693-4c38-9d69-42bbbdcc93a2

FreeCAD_weekly-builds-40655 and 1.0.0.39109 working fine

shaise, 2025-03-30:
Fasteners does not control the property view. Lets wait and see if its only on fasteners, or other objects.

Syres916, 2025-03-30:
This is very bizarre, as I get it with 0.21.1:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-Ribbon 1.8.0.3 (Disabled)
  * btl 0.9.9
  * FreeCAD-themes 2025.1.7
  * DynamicData 2.73.0
  * sheetmetal 0.7.22
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.8
  * ChangeAppFont
  * Quetzal 1.3.3
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Curves 0.6.59
  * A2plus 0.4.68
  * ThreadProfile 1.93.0
```

`App::PropertyLength` is used by `ArchWall` for the wall length and that seems to be OK, I'm struggling with time tonight but tomorrow I'll look at some debugging of the `PropertyItem` `onChange`.

fra589, 2025-03-31:
Hi,
Same behavior with FreeCAD 1.0 and Fastener updated today...
The problem appear just after PR(#432) "Add minimum length option in preferences"...

I think the problem come from changes in FastenersCmd.py.

I'm currently searching in this file :-)

@++;
Gauthier.

shaise, 2025-03-31:
Yes, this is definitely the reason. taking a look

fra589, 2025-03-31:
The problem come from line 701 of FastenersCmd.py

Should be:
    if Units.Quantity(l) < minimumLength.getValueAs(1.0,1):

In place of:
    if Units.Quantity(l).UserString < minimumLength.UserString:

I make the pull request...

@++;
Gauthier.

shaise, 2025-03-31:
@fra589  there are several issues with this code, so i fixed it as well, you can close the PR
@tarman3 please update and check.

fra589, 2025-03-31:
OK... PR closed

Syres916, 2025-03-31:
Humble apologies everyone, @shaise thanks for the fixes.

tarman3, 2025-03-31:
Confirm

closed 2025-03-31 by tarman3