shaise/FreeCAD_FastenersWB #438: Workbench won't load

3nt3, 2025-03-30:
```
17:55:06  During initialization the error "cannot import name 'languagePath' from 'FSutils' (/home/ente/.local/share/FreeCAD/Mod/FreeCAD_FastenersWB/./FSutils.py)" occurred in /home/ente/.local/share/FreeCAD/Mod/fasteners/./InitGui.py
17:55:06  Please look into the log file for further information
17:55:06  During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in /home/ente/.local/share/FreeCAD/Mod/FreeCAD_FastenersWB/./InitGui.py
17:55:06  Please look into the log file for further information`

```

I'm running FreeCad 1.0.0 on arch linux and the Fasteners workbench version 0.5.37

shaise, 2025-03-30:
Sometimes the update does not work correctly. try uninstall and reinstall.

3nt3, 2025-03-30:
tried that already after reading a different issue :(

shaise, 2025-03-30:
It works for me on Ubuntu. From the error message it seems the files are not updated. Do you know where freecad stores its external workbenches on arch?

shaise, 2025-03-30:
Oh - i see from the error message
`/home/ente/.local/share/FreeCAD/Mod/FreeCAD_FastenersWB/`
can you zip and attach FSutils.py from that directory?

3nt3, 2025-07-23:
Just stumbled upon my own thread after googling the issue after I forgot I already had this issue before.

[FSutils.py.tar.gz](https://github.com/user-attachments/files/21396051/FSutils.py.tar.gz)

Still no fasteners for me :(

3nt3, 2025-07-23:
Just updating to `origin/master` seems to work for now

closed 2025-08-08 by luzpaz