shaise/FreeCAD_FastenersWB #455: Bug of with the ST2.9 diameter

shaise, 2025-08-10:
Created by @afabilli and moved here

### Describe your issue

I think I found a bag in the Fasteners add-on. When adding an ISO 7049-C screw and selecting the diameter in ST2.9,
the following message appears in the "Report View" window:

[image]

In the "Tree View" window, the component is marked as incorrect. I tried with other diameters of the same screw type, and the error doesn't appear. The error only occurs with the ST2.9 diameter.

The Fasteners version is updated to V0.5.39 and the FreeCAD version is:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4
  * fasteners 0.5.39
  * Help 1.0.3
  * kicadStepUpMod 11.7.3

Best regards
Alessandro

### FreeCAD version info + SheetMetal WB version

```shell

```

shaise, 2025-08-11:
@afabilli, it should be fixed now.

closed 2025-08-11 by shaise

afabilli, 2025-08-11:
Ok.
@luzpaz Which is the updated version? Because the bug is still present in V0.5.40.

shaise, 2025-08-11:
update to 5.41 beta. if it does not update, uninstall and reinstall